Northeastern-Electric-Racing/Shepherd-BMS #9: Send all Cell Data over CAN

dyldonahue, 2023-09-03:
Another ticket that was investigated for Shepherd 1.0, but never completed. See the info [here (PR),](https://github.com/Northeastern-Electric-Racing/shepherd_bms/pull/165) and [here (ticket), ](https://github.com/Northeastern-Electric-Racing/shepherd_bms/issues/158) Mainly, we need to address the issue explained by @mccauleyma in the comments of that ticket, but a lot of the work may already be done in that PR

dyldonahue, 2024-02-29:
I’m not going to close this _yet_, but i dont think this is a really necessary. Not worth the overhead of CAN spam, but open to different opinions

mccauleyma, 2024-03-01:
@dyldonahue, if time allows, I'd recommend still matching the Orions ability to do this. The way it was done there was this was an extra setting you could enable during firmware upload and is _not_ recommend for normal day to day use. It's a diagnostic tool for debugging the battery pack. Definitely nice to have _if we have time to spare_.

dyldonahue, 2024-03-01:
That makes sense. If remote over IP works, we would be able to achieve this by just reading serial data remotely, but this shouldn't be hard to do so I will plan to get it done

mccauleyma, 2024-03-01:
If you'd prefer that method that works for me. As long as it can be recorded so we can export to MATLAB or whatever we need to analyze

mccauleyma, 2024-03-01:
I guess maybe the argument for CAN is being viewable with our telemetry viewing software. But not a big deal. Any situation that we're looking at cell by cell data we are probably looking at a big issue that we need to manually analyze anyways

dyldonahue, 2024-03-01:
Ok nah I was off base with what we wanted to do with this, if we wanna export it for longer term control can is best. There will be plenty of time to do this so it should be fine

jr1221, 2024-08-06:
@Sabramz I can take care of this after #108 .

closed 2025-01-25 by jr1221