Northeastern-Electric-Racing/Shepherd-BMS #30: Create "Charger CAN" fault

dyldonahue, 2023-09-24:
This is a fault that Orion BMS (the old, prebuilt one we used before Shepherd existed) had implemented, that we haven't yet added. 

Investigate the conditions that this fault refers to, estimate the thresholds that lead to a fault, and add this fault to our faulting index and ensure its functionality aligns with how our fault functions operate

Note: depending on which fault, checking in with Tractive system might be valuable in order to determine fault conditions. Otherwise, let me know if you are having trouble finding any resources to define the conditions or thresholds.

Sabramz, 2023-10-07:
I can't find anything on the interwebz that even alludes to this fault existing.

Sabramz, 2023-10-08:
Our charger speaks I2Cnese.

closed 2023-10-08 by Sabramz