Northeastern-Electric-Racing/Shepherd-BMS #47: Update Thermistor Map to C

nwdepatie, 2023-10-09:
When we were porting all the code over to C, the relevant thermistor map requires a bit of a change to get working. Before, we had it as a C++ vector, which has some methods associated with it that we use. Basically we just need to make this thermistor map work again using what we have set up.

Originally, the thermistor map only had therm numbers that were relevant, but for C we needed to make the array square, and we filled up the rest of the array with `0xFF`, so we need to iterate until we hit `0xFF` and then we can move to the next cell (for averaging the temperatures)

dyldonahue, 2023-10-25:
@nwdepatie to clarify here, Is the map already udpated with 0xff and the methods need to change to accomdate this, or the opposite, or are neither done?

dyldonahue, 2024-01-09:
closing because duplcaite

closed 2024-01-09 by dyldonahue