Northeastern-Electric-Racing/Shepherd-BMS #60: Create Renode files for Shepherd

nwdepatie, 2023-11-11:
Since shepherd builds now (whoo!) we can now start on integrating some development practices we've been honing on Cerberus, specifically Renode.

This ticket involves copying over a lot of the testing and scripts required for Renode to actually work in Shepherd to unit test code. I think that it involves the `cerberus.resc` file and essentially the whole `Testing/` directory in Cerberus. Good endpoint for this ticket is to be able to run a compiled binary of Shepherd in Renode

nwdepatie, 2023-11-11:
Don't worry about if it crashes or not, being able to load it period is where this ticket ends

dyldonahue, 2024-03-28:
closing bc recode is out of scope right now, will investigate this in the summeer

closed 2024-03-28 by dyldonahue