Northeastern-Electric-Racing/Shepherd-BMS #121: Create Accurate SoC percent calculations

jr1221, 2024-09-14:
### Description

Create accurate SoC calcs using online equations for our cells.  Should be done in a way easily transferable to 24A.

### Acceptance Criteria

SoC is accurate.

### Proposed Solution

Use a polynomial approximation or something based on this discharge curve per cell, and then somehow combine that to get overall SoC (probably use low cell?).

jr1221, 2024-09-14:
We have calcs they just don't work.  Let's fix them

dyldonahue, 2024-09-23:
to expand, we have a "curve" that linearly extrapolates soc from a few values. This either doesn't work right, or the curve values do not produce an accurate voltage to SoC conversion, or a combination of both

jr1221, 2024-10-06:
We are using 21700 batteries.

closed 2025-03-24 by jr1221